ndabas/pico-setup-windows #2: "Visual Studio Code for Pico" doesn't work + vscode has CMake problem

BlueDig, 2021-02-12:
I am quite old to programming but quite new to vscode, so maybe I am missing something obvious not mentioned in the readme file. Two problems:
- "Visual Studio Code for Pico" does not work at all - some shadow blinks on the screen for a fraction of second - looks like some window opens then closes immediately.
- If I start vscode from "Developer command prompt for Pico", the vscode complains about CMake:
[main] Unable to determine what CMake generator to use. Please install or configure a preferred generator, or update settings.json, your Kit configuration or PATH variable. Error: No usable generator found.

ndabas, 2021-02-13:
You need to configure the CMake extension as explained in the [Getting started guide](https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf), section 8.2.4. You will need to set the CMake Generator to "NMake Makefiles".

I'm not sure why the "Visual Studio Code for Pico" shortcut did not work for you, given that you do have Visual Studio Code installed. I'll probably add a check there to stop and show an error message if it fails.

BlueDig, 2021-02-13:
Thanks for the help- There is still something wrong with the paths, however: the path to nmake.exe is not set. I did not use the Microsoft build tool on this particular computer before, so the build tools were installed by your package, but the path was not set.

ndabas, 2021-02-14:
I really should've asked at the outset -- what version of Windows are you using - 7 or 10? 32 or 64 bit? I haven't tested extensively outside of Windows 10 x64, so there might be some issues when installing on other versions.

When you open the "Developer command prompt for Pico", do you see any error messages in the terminal window?

BlueDig, 2021-02-15:
Win 10 x64 20H2. No error messages in Developer Command prompt, but "Visual Studio Code for Pico" doesn't work.
After many problems with Cmake project configuration I finally managed to get the vscode working correctly with all the Pico stuff. Simple Blink compiles to 24 KiB binaries; disassembly shows dozens of unnecessary routines, never called code getting into this binary - but this is a different story.

ndabas, 2021-02-15:
Thanks for the update. I don't see any further action here apart from adding some more error checking and logging in the installer. Closing.

closed 2021-02-15 by ndabas