ndabas/pico-setup-windows #3: Error during the "Running post-transaction hooks" process

norfanos, 2021-02-13:
I'll see what I can do on my end, but here is the issue:

```
:: Running post-transaction hooks...
(1/1) Updating the info directory file...
Remove-Item : Cannot find path 'C:\Users\me\Downloads\pico-setup-windows-master\build\openocd' because it does not exist.
At C:\Users\me\Downloads\pico-setup-windows-master\build.ps1:80 char:3
+   Remove-Item .\build\openocd -Recurse -Force
+   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (C:\Users\norfa\...r\build\openocd:String) [Remove-Item], ItemNotFoundException
    + FullyQualifiedErrorId : PathNotFound,Microsoft.PowerShell.Commands.RemoveItemCommand
```

norfanos, 2021-02-13:
I manually downloaded the openocd-rp2040 from their GIT and unzipped it into the build directory, renamed it to match your script as "openocd" and re-ran the script and it is now continuing. I forgot to mention I was running the x64 script. I checked to see if there were any other errors in the script that would have prevented the openocd package and there were none. Thanks for getting this together, and hope the workaround might help someone until the process is updated.

ndabas, 2021-02-13:
Hi, thank you for reporting this. The PowerShell script is basically trying to delete a non-existent folder (which would happen on the first run), and erroring out. I'll fix this by adding a check there.

commit 8e00c680 (2021-02-15): Don't try to delete directory if it doesn't exist

Fixes #3.
closed 2021-02-15 by ndabas