ndabas/pico-setup-windows #4: Ninja needs to be updated to 1.10.2

eriklundh, 2021-03-03:
I experienced a problem with path handling in the version of ninja (1.8.2) that was installed by pico-setup-windows release 0.3.0
I found the issue already reported on the Github issue tracker for ninja. 
I downloaded ninja 1.10.2 from https://github.com/ninja-build/ninja/releases, replaced the ninja.exe binary with the 1.10.2 version, and that solved the path problem

ndabas, 2021-03-04:
We don't bundle ninja explicitly, I guess it came with the CMake installer.

I plan to put out new releases of the installer every couple of weeks or so with the latest releases of the bundled software, so when CMake updates _their_ bundled copy of ninja, this installer will get the update too.

Closing this as wontfix for now because `pico-sdk` does not use ninja on Windows anyway, they use NMake Makefiles.

closed 2021-03-04 by ndabas

eriklundh, 2021-03-04:
> We don't bundle ninja explicitly, I guess it came with the CMake installer.
>
> I plan to put out new releases of the installer every couple of weeks or so with the latest releases of the bundled [...]

Sorry if I was unclear.
In several cases when you use your installer on a blank Windows machine, you end up with ninja instead of nmake.
I haven't dug into the reasons why. But I don't think it is because of the installer per se: I googled the ninja 1.8.2 error and found some discussion around an issue at raspberrypi/pico-sdk  where Alasdair Allan (of Pi Trading) expressed confusion that ninja was invoked in a reply to a user that apparently had done a manual Windows install.
My simple fix was to find a working version of ninja: 1.10.2
I humbly suggest that you add a download and patch of the 1.10.2 version (a zip file that need to be unzipped and target the right place in the tool tree)
I really appreciate your effort with the installer, especially since I have a long history with PI and some of the folks behind it.

/Erik

aallan, 2021-03-04:
> …found some discussion around an issue at raspberrypi/pico-sdk where Alasdair Allan (of Pi Trading) expressed confusion that ninja was invoked in a reply to a user that apparently had done a manual Windows install.

I don't think confusion is the right word there. 

CMake Tools plugin had an update shortly before the release of Raspberry Pi Pico and `ninja` rather than `nmake` was made the default generator. We haven't tested with` ninja`, and aren't using it internally. We therefore still recommend `nmake` as the generator when using the SDK on Windows. We don't anticipate either testing with, or changing to recommending `ninja`.

Then click on the Cog Wheel at the bottom of the navigation bar on the left-hand side of the interface and select "Settings". Then in the Settings pane click on "Extensions" and the "CMake Tools configuration" scroll down to "Cmake: Generator" and enter "NMake Makefiles" into the box to switch over.

eriklundh, 2021-03-04:
> I don't think confusion is the right word there.
>
> CMake Tools plugin had an update shortly before the release of Raspberry Pi Pico and `ninja` rather than `nmake` was made the default generator. We [...]

Alasdair,
Sorry if I misrepresented you comment on that issue, it was just how I recalled it.
Thanks for the configuration tip: 
However, ninja 1.10.2 now works just fine on Windows.
BTW: Where you by any chance at the PI office (at the science park) on Dec 13/2019? (pre-pandemic time warp)
Something about the hedon Saint Lucia traditions of Sweden that we used to scare Nobel Prize winners with?
/Erik

ndabas, 2021-03-04:
This is documented in the [Getting Started guide](https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf). You will need to use either `cmake -G "NMake Makefiles"` or use the VS Code settings to configure the default generator, like @aallan has kindly re-iterated here.

That said, to address your issue:

> In several cases when you use your installer on a blank Windows machine, you end up with ninja instead of nmake.

Do you mean that `nmake` did not get installed at all? It's installed as part of the Visual Studio Build Tools. I am aware that the installation fails in certain cases (e.g. if you do not have at least .NET Framework 4.6 installed) -- I am working on adding more error checks with every release, and the next one should have a check for this as well.

eriklundh, 2021-03-04:
> This is documented in the [Getting Started guide](https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf). You will need to use either `cmake -G "NMake Makefiles"` or use the VS Code [...]

I have done manual install on one other machine.
I welcomed the automatic install, since we now are doing RP2040 product development. 
I did not actively chosen Ninja, but welcomed the change, since I have bad experiences of nmake from the early days of Microsoft C, in the days before Visual C, in the 1980s when MS internally  used unix(actually their own xenix) to cross-compile products since dos/Windows could not self-host the toolchain for larger compiles.  Back in the day, nmake was full of its own oddities that separated it from the Unix make, since MS culture at the time encouraged 'do our own thing' and not-invented-here. In the end we lost so much time to MS C tool quirks that later I famously (and naively) declined a visit from Bill Gates w entourage at a big trade show in 1987 where we exhibited our first mass product, just because we had switched to the quick-full-cycle iteration-encouraging TurboC, and young as we was, we thought that Borland compilers would dominate the world. for the foreseeable future  Little did we know. Of course Cmake is a metatool that takes care of the oddities of nmake, ninja and other make tools. It was just that my old scepsis, not of MS of today, but the oddities of the olds tools, made me welcome an alternative. Some of my old clients embraced ninja for really large builds when it first appeared, and I have heard mostly good things about it.

aallan, 2021-03-04:
> BTW: Where you by any chance at the PI office (at the science park) on Dec 13/2019? (pre-pandemic time warp)

Nope, sorry. Checking my calendar I think I was in San Jose that week.

eriklundh, 2021-03-04:
> This is documented in the [Getting Started guide](https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf). You will need to use either `cmake -G "NMake Makefiles"` or use the VS Code [...]

Nmake was installed by pico-windows-setup 0.3.0 on a fresh Windows 10.
It is still a good idea to validate the toolchain.

Kabron287, 2021-12-11:
Windows 10 x64
VS Code 
Cmake: Generator Nmake Makefiles
Checking "GNU Arm Embedded Toolchain"...
Checking "CMake"...
Checking "Visual Studio"...
Checking "Python 3"...
Checking "Git"...

 **[cmake] CMake Error: Could not create named generator Nmake Makefiles**

Any help?

ndabas, 2021-12-11:
@Kabron287 It's `NMake Makefiles` not `Nmake Makefiles` (note the capitalization.)

Kabron287, 2021-12-11:
Sure! Thanks!