ndabas/pico-setup-windows #5: pico-setup.cmd runs in Command Prompt

santaimpersonator, 2021-03-19:
I am using the release installation executable [pico-setup-windows-0.3-x64.exe] from the v0.3 release and at the end of the installation completion, a prompt allows users to have the executable _Clone and build Pico repos_.

[image]

I believe this option calls the `pico-setup.cmd` script; however, in my case, it appears to run through the default **Command Prompt**, which results in errors.

[image]

Instead, I think the `pico-setup.cmd` script is intended to:

- run through the **Developer Command Prompt for VS2019**
- run through the **Developer Command Prompt for Pico** *(w/ shortcut link)*

Otherwise, I think another option is if the `pico-env.cmd` script was called in the **Command Prompt** prior to executing the `pico-setup.cmd` script.

**Computer Platform:**
> Edition	Windows 10 Pro
> Version	20H2
> OS build	19042.867

ndabas, 2021-03-21:
Hi,

Thank you for the very detailed bug report!

The first line in `pico-setup.cmd` calls `pico-env.cmd`, and that part is clearly working in your case because we can see output from both CMake and nmake.

I think what is happening is that there are spaces or other special characters in the path (the redacted download location in your screenshots), and I don't recall testing the scripts with such paths. I'm probably missing some quotation marks in the `pico-*.cmd` scripts.

If that is the case -- I'm working on another release, and I will test with such paths for that one.

santaimpersonator, 2021-03-22:
Hey @ndabas

I definitely forgot that `pico-setup.cmd` already calls `pico-env.cmd`. My file path is just `C:\Users\<username>\Documents\Pico` and my username is mostly just a bunch of letters and numbers; the only special character is a `.` (period).

I'll be honest... I am not sure why/how, but the setup execution seems to work for me now.

1. The only thing I did was test `pico-setup.cmd` with `call "pico-env.cmd"` instead. The change worked; for some reason, the command prompt was having issues executing `call "%~dp0pico-env.cmd"` _(kind of weird)_.
2. Then I removed my `Pico` folder.
3. Then I re-ran the v0.3 executable again and didn't opt to _Clone and build Pico repos_.
4. Tested the change again, which worked. Then I tried running the original code to verify if that was the problem and to my surprise it worked, unlike the several dozen times before.
5. I deleted my `Pico` folder and re-ran the v0.3 executable _( with `Clone and build Pico repos` selected)_, a couple more times. Everything seems  to be working like it should.

The only thing that I can see, which is different is that this shows up in the command prompt before `pico-env.cmd` is called properly:
> `Refreshing environment variables from registry for cmd.exe. Please wait...Finished..`
> `C:\Users\<username>\Documents\Pico>call "C:\Users\<username>\Documents\Pico\pico-env.cmd"`

I didn't change any file paths, system variables, or reset my computer. I am kind of at a loss to why everything suddenly started working, after I had been struggling with this all week... LOL. Unless you have any ideas further into the problem, I think we can probably close the issue.

ndabas, 2021-04-03:
I don't really have any further ideas on this, just that we could do with some more error checks in the scripts. That's on my to-do anyway, so closing this issue now.

closed 2021-04-03 by ndabas