ndabas/pico-setup-windows #6: Shortcuts not working.. Project generator and VSC

Thiadeth, 2021-03-21:
hi, im quite new but the Shortcut CMD commands aren't working for me, due to the double ampersand && in both the project generator and VSC for pico. removing the double ampersand works though, unless it modifies the desired output

Project generator
> `C:\WINDOWS\system32\cmd.exe cmd /c call "C:\Users\_username_\Desktop\Pico\pico-env.cmd" &&python "C:\Users\_username_\Desktop\Pico\pico-project-generator\pico_project.py" --gui`
>  
>  Start in : `C:\Users\_username_\Desktop\Pico\pico-project-generator`

VSC for pico 
> `C:\WINDOWS\system32\cmd.exe /c call "C:\Users\_username_\Desktop\Pico\pico-env.cmd" &&code`
> 
> Start in : `C:\Users\_username_\Desktop\Pico`

[image]

ndabas, 2021-04-01:
I'm not sure how removing the double ampersand would work. It's just saying "run this command when the last one succeeds." Did you replace the double ampersands with single ampersands?

ndabas, 2021-06-14:
Closing because I haven't heard back from the issue author.

closed 2021-06-14 by ndabas