ndabas/pico-setup-windows #7: Error C1083 in only PIO project generated with the Code Generator // Problems to access 'pico/stdlib.h': No such file or directory

cmarcante, 2021-04-02:
Hello,
Please, I am a new programmer under VS Code.
On a windows 10 computer (Spanish) I was able to compile manually all the examples but I tried the code generator with new projects under VSC (Windows 64) and I always got errors, when I compiled the first generated example code. The example code it is very usefully.

I always run the VSC from "Visual Studio Code for Pico" access. I tried the Raspberry Pi environment configuration from a PDF instructions file and it failure  in compile the example generated with Code Generator.

Now I selected the easiest example, with only PIO and debug with serial interface but I could not compile the generated code.

I received always error. The error is "C1083: Cannot open file include: 'pico / stdlib.h': No such file or directory". 
The   hardware/divider.h library generated errors too.  

Test: I added to CMakeList the stdlib but it did not solved the problem.
I added to CMakeList the hardware_divider and it clean the warning but I got other errors.

          # Add any user requested libraries
          target_link_libraries(cm04_rfi01
                  pico_stdlib
                  hardware_pio
                  hardware_divider
                  )

Please, I tried many options during days; I cannot confirm where the problem is. 

Thanks

[CMakeLists.txt](https://github.com/ndabas/pico-setup-windows/files/6250272/CMakeLists.txt)
[pico_PIO_With_Code_generator.zip](https://github.com/ndabas/pico-setup-windows/files/6250276/pico_PIO_With_Code_generator.zip)

ndabas, 2021-04-03:
It looks like you were able to get the development tools installed and running with no issue, since you mention that you got the examples compiled.

Please post your issue in the [Raspberry Pi Pico Forum](https://www.raspberrypi.org/forums/viewforum.php?f=144), there is a good chance somebody will be able to help you over there.

Closing since this is not an issue with this project.

closed 2021-04-03 by ndabas

cmarcante, 2021-04-03:
Hello, 
Thanks for your suggestion.
I will uninstall all and I install again all the C compiler plus VSC.
Regards

ndabas, 2021-04-03:
I don't think reinstalling will help here. It seems your dev toolchain is fine to me. Please post the details of the issue on the forum like I suggested, and I'm sure you will get some helpful hints there.