ndabas/pico-setup-windows #9: Git/bash doesn't work anymore after running this installation

zapta, 2021-06-02:
After running this installation, my git bash doesn't work anymore. When I opens it I get a windows cmd shell rather than bash which I used to get before the installation.

Any suggestion?

zapta, 2021-06-02:
This is the git/bash I was using. https://gitforwindows.org/

ndabas, 2021-06-02:
Do you mean you get the Windows default console host rather than the MinTTY terminal emulator? You should still get a bash prompt in that window rather than cmd.exe.

If that is the case, then yes, this is because pico-setup-windows also installs Git for Windows, which will upgrade your install with the included version and the supplied options. The supplied options are, in this case, from the [Getting Started with Raspberry Pi Pico guide](https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf), section 9.2.1.5 Installing Git:

> Ensure you tick the checkbox to allow Git to be used from third-party tools and, unless you have a strong reason otherwise, when installing Git you should also check the box "Checkout as is, commit as-is", select "Use Windows' default console window", and "Enable experimental support for pseudo consoles" during the installation process.

The option that might be causing this is the "Use Windows' default console window", which is specified in https://github.com/ndabas/pico-setup-windows/blob/master/git.inf. If this is correct:
- You can 'fix' your installation by just re-running the Git for Windows installer again, manually selecting the options you need.
- If you'd like to contribute this change, you can edit the git.inf file and test if the installation and Pico SDK build works fine, and submit a PR with that change.

zapta, 2021-06-02:
Thanks Nikhill, your analysis is correct. I got bash running in the cmd.exe
window. Reinstalling git bash fixed it.

I also had a general problem using the SDK.  I was able to build examples
from github but the upload button didn't work and instead of uploading the
uf2 file to my board and running it, I got a Windows dialog asking me what
program to use to handle an .elf file.

You can see more details here
https://www.raspberrypi.org/forums/viewtopic.php?f=145&t=313069

ndabas, 2021-06-02:
You're welcome, glad to have been of help! I'm closing this issue now as there's no further action.

closed 2021-06-02 by ndabas