ndabas/pico-setup-windows #10: CMAKE errors

rkayakr, 2021-06-06:
CMake Error at C:/Users/rkaya/Documents/Pico/pico-sdk/cmake/preload/toolchains/find_compiler.cmake:28 (message):
  Compiler 'arm-none-eabi-gcc' not found, you can specify search path with
  "PICO_TOOLCHAIN_PATH".

CMake Error: CMake was unable to find a build program corresponding to "NMake Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_ASM_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
Building blink
The system cannot find the path specified.
'nmake' is not recognized as an internal or external command,
operable program or batch file.

ndabas, 2021-06-06:
Could you please be more descriptive about what the issue really is? What are the steps you followed to reproduce this problem etc.

Did you install all of the included software? Did you use the "Developer Command Prompt for Pico" shortcut to open a command prompt? What version of Windows do you have?

rkayakr, 2021-06-06:
Thanks for the reply. I ran the script  and installed all software on my Windows 10 laptop. I copy and pasted the error messages I saw at the end. I  reran this snippet
C:\Users\pico\Downloads> cd pico-examples
C:\Users\pico\Downloads\pico-examples> mkdir build
C:\Users\pico\Downloads\pico-examples> cd build
C:\Users\pico\Downloads\pico-examples\build> cmake -G "NMake Makefiles" ..
C:\Users\pico\Downloads\pico-examples\build> nmake

to reproduce and got the same result.

I don't see a "Developer Command Prompt for Pico",  just "Developer Command Prompt for VS 2019"

   Bob

ndabas, 2021-06-13:
I think all you need to do to fix this is to run these commands from the "Developer Command Prompt for Pico" shortcut that should be installed in the directory that you chose during setup, which is C:\Users\<username>\Documents\Pico by default.

Otherwise, running these from "Developer Command Prompt for VS 2019" should also work.

rkayakr, 2021-06-14:
Thanks. You are correct.
I was looking in the Windows start menu so I didn't see the "Developer
Command Prompt for Pico" shortcut, just the "Developer Command Prompt for
VS 2019" .
     Bob

ndabas, 2021-06-14:
Great, glad I could help!

closed 2021-06-14 by ndabas