ndabas/pico-setup-windows #11: C++ project

Trance-Paradox, 2021-06-14:
After creating the project with ***Pico Project Generator*** and selecting all C++ options under ***Code Options*** I opened up the project on vscode from pico console. When I build the project in release mode it does not generate uf2 or even the hex file.

[image]

ndabas, 2021-06-14:
It seems that CMake has generated the build files but did not compile them. On the command line, I would do this:

```
cd build
cmake -G "NMake Makefiles" ..
nmake
```

Unfortunately I don't have much experience with configuring VS Code properly to do this -- I suggest that you ask in the [forum](https://www.raspberrypi.org/forums/viewforum.php?f=144), and if you feel you have found a bug in the generated configuration -- you can open an issue in the [pico-project-generator](https://github.com/raspberrypi/pico-project-generator) repo.

It looks like you have been able to install the environment properly -- so this isn't an issue with this project (pico-setup-windows).

Trance-Paradox, 2021-06-14:
> It seems that CMake has generated the build files but did not compile them. On the command line, I would do this:
>
> ```
> cd build
> cmake -G "NMake Makefiles" ..
> nmake
> ```
>
> Unfortunately I don't have [...]

Seems they only currently support c projects. I will try to create an issue in the project creator

ndabas, 2021-06-14:
Closing as there is no further action in pico-setup-windows.

closed 2021-06-14 by ndabas