ndabas/pico-setup-windows #12: "Visual Studio Code for Pico" and "Pico Project Generator" shortcuts not working

LiterallyKey, 2021-07-06:
I am probably missing something obvious or being stupid as I am very new to all of this, but it seems that the two core shortcuts that would allow me to use my picos are broken. I saw that some other people have had similar issues, but I didn't see any solution (closest was someone saying that they fiddled with it until it worked). What can I do to get this to work/be able to actually use my picos?

ndabas, 2021-07-07:
Can you describe in some detail what this means exactly:
>  it seems that the two core shortcuts that would allow me to use my picos are broken

I mean, what happens when you double-click on those shortcuts?

LiterallyKey, 2021-07-08:
Something that looks like a command prompt window flashes for a fraction of a second (to the point where it doesn't even get fully opaque) and then nothing happens.

ndabas, 2021-07-08:
Would you be able to run a few commands to help debug this issue?

Open a command prompt, `cd` to the installation directory (which is `"%USERPROFILE%\Documents\Pico"` by default), and then run:

```cmd
cmd.exe /c call pico-env.cmd && code

cd pico-project-generator
cmd.exe /c call pico-env.cmd && python pico_project.py --gui
```

Do these commands launch Visual Studio Code and the Pico Project Generator for you?

I'd appreciate it if you could paste the output from those commands here.

LiterallyKey, 2021-07-09:
Putting in the whole block at once gave the following output and did not open anything:

PICO_sdk_PATH=C:\Users\danie\OneDrive\Documents\Pico\pico-sdk`
PICO_examples_PATH=C:\Users\danie\OneDrive\Documents\Pico\pico-examples
PICO_extras_PATH=C:\Users\danie\OneDrive\Documents\Pico\pico-extras
PICO_playground_PATH=C:\Users\danie\OneDrive\Documents\Pico\pico-playground
OPENOCD_SCRIPTS=C:\Users\danie\OneDrive\Documents\Pico\tools\openocd-picoprobe\scripts

C:\Users\danie\OneDrive\Documents\Pico>cmd.exe /c call pico-env.cmd && code
PICO_sdk_PATH=C:\Users\danie\OneDrive\Documents\Pico\pico-sdk
PICO_examples_PATH=C:\Users\danie\OneDrive\Documents\Pico\pico-examples
PICO_extras_PATH=C:\Users\danie\OneDrive\Documents\Pico\pico-extras
PICO_playground_PATH=C:\Users\danie\OneDrive\Documents\Pico\pico-playground
OPENOCD_SCRIPTS=C:\Users\danie\OneDrive\Documents\Pico\tools\openocd-picoprobe\scripts

C:\Users\danie\OneDrive\Documents\Pico>
C:\Users\danie\OneDrive\Documents\Pico>cd pico-project-generator

C:\Users\danie\OneDrive\Documents\Pico\pico-project-generator>cmd.exe /c call pico-env.cmd && python pico_project.py --gui

ndabas, 2021-07-10:
Thanks. It looks like you either do not have Visual Studio 2019 installed, or that the pico-env.cmd script is unable to find your VS installation. If you believe that you do, in fact, have installed Visual Studio, can you try running this command to list your VS installs?

```cmd
"C:\Program Files (x86)\Microsoft Visual Studio\Installer\vswhere.exe" -products *
```

If you are unable to find `vswhere.exe` locally on your system, you can get it from: https://github.com/microsoft/vswhere/releases

On a related note, I have already added some more checks for this sort of thing to the installer code, and based on your inputs, I will probably add even more checks for Visual Studio. I haven't put out a release with those updates yet because I wasn't sure about the actual issues users have with Visual Studio installations.

LiterallyKey, 2021-07-12:
Strangely enough it just started working. I forgot I had pinned the shortcut to open vs code on my taskbar instead of the terminal, but when I clicked on it it actually worked. I have no idea what I did, but I will test it and make sure it is truly working and check back.

LiterallyKey, 2021-07-12:
Now the the examples don't even let me debug them. It seems that it is not finding pico/stdlib.h (cannot open source file "pico/stdlib.h") as well as an include error on the same line (#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit (C:\Users\danie\OneDrive\Documents\Pico\pico-examples\hello_world\serial\hello_serial.c). In the seven segment display example it is also giving this error: CMake Error at CMakeLists.txt:9 (pico_add_extra_outputs):Unknown CMake command "pico_add_extra_outputs".. 

It is promising that visual studio now launches from the shortcut, but I am not sure what is wrong now. 

Also, the return for the command you said to run:

Microsoft Windows [Version 10.0.19042.1083]
(c) Microsoft Corporation. All rights reserved.

C:\Users\danie>"C:\Program Files (x86)\Microsoft Visual Studio\Installer\vswhere.exe" -products *
Visual Studio Locator version 2.7.1+180c706d56 [query version 2.7.3111.17308]
Copyright (C) Microsoft Corporation. All rights reserved.

instanceId: 9fb930e7
installDate: 7/6/2021 1:17:55 PM
installationName: VisualStudio/16.10.3+31424.327
installationPath: C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools
installationVersion: 16.10.31424.327
productId: Microsoft.VisualStudio.Product.BuildTools
productPath: C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\Common7\Tools\LaunchDevCmd.bat
state: 4294967295
isComplete: 1
isLaunchable: 1
isPrerelease: 0
isRebootRequired: 0
displayName: Visual Studio Build Tools 2019
description: The Visual Studio Build Tools allows you to build native and managed MSBuild-based applications without requiring the Visual Studio IDE. There are options to install the Visual C++ compilers and libraries, MFC, ATL, and C++/CLI support.
channelId: VisualStudio.16.Release
channelUri: https://aka.ms/vs/16/release/channel
enginePath: C:\Program Files (x86)\Microsoft Visual Studio\Installer\resources\app\ServiceHub\Services\Microsoft.VisualStudio.Setup.Service
releaseNotes: https://docs.microsoft.com/en-us/visualstudio/releases/2019/release-notes-v16.10#16.10.3
thirdPartyNotices: https://go.microsoft.com/fwlink/?LinkId=660909
updateDate: 2021-07-06T17:17:55.0897384Z
catalog_buildBranch: d16.10
catalog_buildVersion: 16.10.31424.327
catalog_id: VisualStudio/16.10.3+31424.327
catalog_localBuild: build-lab
catalog_manifestName: VisualStudio
catalog_manifestType: installer
catalog_productDisplayVersion: 16.10.3
catalog_productLine: Dev16
catalog_productLineVersion: 2019
catalog_productMilestone: RTW
catalog_productMilestoneIsPreRelease: False
catalog_productName: Visual Studio
catalog_productPatchVersion: 3
catalog_productPreReleaseMilestoneSuffix: 1.0
catalog_productSemanticVersion: 16.10.3+31424.327
catalog_requiredEngineVersion: 2.10.2174.31177
properties_campaignId:
properties_channelManifestId: VisualStudio.16.Release/16.10.3+31424.327
properties_nickname:
properties_setupEngineFilePath: C:\Program Files (x86)\Microsoft Visual Studio\Installer\setup.exe

C:\Users\danie>

ndabas, 2021-07-13:
Well. Thanks for that, at least it seems that your tools are working now. I don't know what the issue is though if it just started working on its own.

Regarding the build/debug issues, you can probably ask in the [Pico forum](https://www.raspberrypi.org/forums/viewforum.php?f=144), there are plenty of knowledgeable people there with deep knowledge of the SDK (including the authors.)

I'm closing this issue for now; please feel free to re-open it if there is still a problem with the tools installation or if you have any more info to share.

closed 2021-07-13 by ndabas