ndabas/pico-setup-windows #15: ERROR: "GNU Arm Embedded Toolchain" is required but was not found.

LeifBloomquist, 2021-09-13:
Hi, just downloaded and ran v0.3.3 .   When I click on "Developer Command Prompt for Pico"  (or run any of the other setup scripts), I get:

```
OPENOCD_SCRIPTS=C:\Pico\tools\openocd-picoprobe\scripts
**********************************************************************
** Visual Studio 2019 Developer Command Prompt v16.11.2
** Copyright (c) 2021 Microsoft Corporation
**********************************************************************
Checking "GNU Arm Embedded Toolchain"...
ERROR: "GNU Arm Embedded Toolchain" is required but was not found.
Checking "CMake"...
Checking "Visual Studio"...
Checking "Python 3"...
Checking "Git"...
```

I also downloaded and installed the ARM tools directly from https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads  , even rebooted, no change.

The script seems to be looking for arm-none-eabi-gcc, which doesn't seem to be on my system.

LeifBloomquist, 2021-09-13:
Adding C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.07\bin  to my PATH manually solved the problem - so far :-)

ndabas, 2021-09-14:
That's interesting. Normally the installer for the ARM compilers adds this to the PATH automatically. Let me do some tests and see if they've changed anything in the newest version. Thanks for reporting this!

ndabas, 2021-11-09:
This seems to work just fine with the newest version of the ARM GCC release. I have not been able to replicate this issue; if anyone sees this issue again and can provide more info, I would be happy to re-open the issue. As it is there is nothing to do, so I am closing the issue.

closed 2021-11-09 by ndabas