ndabas/pico-setup-windows #17: RPi Pico Project Generator - NMAKE -> invalid option 'j'

quizque, 2021-11-24:
When trying to enable the "Run build after generation" feature inside of the Raspberry Pi Pico Project Generator all files are generated fine but when it tries to build, it generates an error and fails.

Heres the options I have selected
[image]

Here is the log of it generating just fine
[image]

This is where the error occurs
[image]

Here is a link to the project it generated
[ProjectName2.zip](https://github.com/ndabas/pico-setup-windows/files/7598407/ProjectName2.zip)

ndabas, 2021-11-25:
This appears to be a bug with the [Pico Project Generator](https://github.com/raspberrypi/pico-project-generator) and not this installer project. Please open an issue there.

A quick look shows that the problem is here: https://github.com/raspberrypi/pico-project-generator/blob/master/pico_project.py#L1179 -- `makeCmd = 'nmake -j ' + str(cpus)` -- `nmake` has no `-j` option. You could actually just submit a PR changing that line to `makeCmd = 'nmake'` which should fix it.

I'm closing this issue for now because there is nothing to do in this project.

closed 2021-11-25 by ndabas

JamesH65, 2021-11-25:
I've fixed this on the pico project generator master branch. Thanks for the diagnosis.

ndabas, 2021-11-25:
@JamesH65 great, thank you!