ndabas/pico-setup-windows #18: Picoprobe not detected.

prime6809, 2022-01-15:
Hi, only installed picoprobe as I already have a working build environment running on WSL2, so only need picoprobe for debugging.

I have installed in : D:\pico (with picoprobe in d:\pico\tools\openocd-picoprobe).

Trying to start picoprobe with : 

openocd -f interface/picoprobe.cfg -f target/rp2040.cfg -s scripts

It fires up but can't detect the picoprobe.

`openocd -f interface/picoprobe.cfg -f target/rp2040.cfg -s scripts
Open On-Chip Debugger 0.10.0+dev-geb22ace-dirty (2021-02-23-23:12)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
Warn : Transport "swd" was already selected
adapter speed: 5000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: Failed to open or find the device
Error: Can't find a picoprobe device! Please check device connections and permissions.`

It does this in both a standard user & administrator cmd window.

The 2 picoprobe interfaces show up in device manager under libusb-win32 devices and are using winusb 1.2.6.0. I've tried with zadig changing the driver on both to WinUSB (v6.1.7600.16385) and to libusb-win32 (1.2.6.0) but the result is the same?

I have previously used the picoprobe with a cygwin compiled openocd but this has recently stopped working. However I'm pretty sure the hardware is not at fault and just to be sure reflashed the picoprobe with the compiled firmware.

Cheers.

Phill.

Terstegge, 2022-02-24:
I don't know if this helps, but I also had problems that my picoprobe was not detected as a USB device. After some experiments with reset circuits I found out that every time I unplug the probe, I have to wait approx. 5-10 seconds before plugging it in again. After waiting, the probe was always recognized. If I didn't wait long enough, the probe was not recognized.....  This also worked wthout any reset delay (I connected RUN to 3V3).

ndabas, 2022-11-17:
This should be much better now as Picoprobe is now just a standard CMSIS-DAP device.

closed 2022-11-17 by ndabas

singhWeb, 2023-08-11:
@prime6809 have you find any solutions to it?