ndabas/pico-setup-windows #21: Not digitally Signed.

jtannenba, 2022-05-02:
This did not want to run on my Windows 11 machine.  Not digitally signed.  Had to set-executionpolicy. A true pain.

.\build.ps1 : File D:\Documents\Pico\pico-setup-windows-master\build.ps1 cannot be loaded. The file
D:\Documents\Pico\pico-setup-windows-master\build.ps1 is not digitally signed. You cannot run this script on the
current system. For more information about running scripts and setting execution policy, see about_Execution_Polic
https:/go.microsoft.com/fwlink/?LinkID=135170

ndabas, 2022-05-03:
Running a well-documented first-party one-line command is "a true pain"? You do not need to run this script as an end user. If you want to do some development work on the project, `Set-ExecutionPolicy` is standard procedure for PowerShell development.

Closing this issue as no further action is needed.

closed 2022-05-03 by ndabas

jtannenba, 2022-05-03:
Not if you are not a powershell user.

jtannenba, 2022-05-05:
Ran in a powershell as admin:
After quite awhile, got this:

Command ' & "$MSYS2Path\usr\bin\bash" -leo pipefail -c "$cmd" ' exited with code 2At D:\Documents\Pico\pico-setup-windows-master\common.ps1:39 char:7+       throw "Command '$cmd' exited with code $LASTEXITCODE"+       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~    + CategoryInfo          : OperationStopped: (Command ' & "$M...ted with code 2:String) [], RuntimeException    + FullyQualifiedErrorId : Command ' & "$MSYS2Path\usr\bin\bash" -leo pipefail -c "$cmd" ' exited with code 2

So, does that mean the whole thing is bad?  Is there a way to recover?
Joe