ndabas/pico-setup-windows #22: VS Code update fails with "Access is denied."

TPMoyer, 2022-06-07:
First off:   Thank you for writing this script.  Is WAY better than the many manual step alternative.

Small foible with the install.   After opening the command prompt which sets in place the environment, using Visual Studio Code, and closing same... Visual Studio tries to peek for and install updates.   This fails with
[image]
 
How can I go in and change permissions
to what series of directories
to allow the updates to propagate through to success?

DanielTomlinsonSage, 2022-07-25:
This issue can be closed.

ndabas, 2022-11-17:
Thanks for documenting the issue and the fix! Closing.

closed 2022-11-17 by ndabas