ndabas/pico-setup-windows #27: Making this work on Windows 10 - 6 Oct 2022

StewBC, 2022-10-06:
Hi,

I had to make the following changes to get this built on Windows 10 as of 6 Oct 2022.

1) Possibly an issue just for me.  I had SED and AWK defined to point at gnu-sed and awk on my system and that caused havoc.  I have the gnu tools installed in C:\Program Files\... and that space between Program and Files wreaks havoc.  In Poweshell I just had to say $env:sed="" and $env:awk="" to clear those settings.  I also edited my path so that it had minimal stuff in it (removed gnu and anything else that wasn't windows, really).

2) x64.json and x86.json - Doxygen version has been updated.  On https://doxygen.nl/download.html I found the link for a binary distribution for windows.  Currently that is https://www.doxygen.nl/files/doxygen-1.9.5-setup.exe.  I just updated the json files withe the new version (I changed both file and href).

3) The "treat all warnings as errors" will stop the compilation of openocd.  Specifically, I had to set -Wno-dangling-pointer -Wno-stringop-overflow -Wno-misleading-indentation.  Just not setting -Werror also works.
For anyone wanting to build as it stands and doesn't know how to "fix" the Makefike that needs fixing (this includes me), what I did was run the .\build.ps1 .\x64.json and when the compile failed, I opened build\openocd\Makefile and found the line GCC_WARNINGS = <stuff>.  I then deleted the "stuff".  I re-ran the build and when the "stuff" got added to line, I removed -Werror from the end of the "stuff" and saved quickly.  That was enough to get me through.  Since I was curious what warnings blocked me, I went through this process several times to identify the specific warnings as noted earlier.

After all this I was able to successfully run the build.ps1 script.

StewBC, 2022-10-06:
Well, _now_ I notice there's a release already.  That's what happens when you are over-eager.

closed 2022-10-06 by StewBC

ndabas, 2022-10-10:
Well, thanks for documenting what you did anyway, because the last release is a bit outdated.