ndabas/pico-setup-windows #29: Possible Update

microenh, 2022-11-15:
The installer is failing when building picoprobe and copying documentation from RaspberryPi.org.

The picoprobe failure is because the picoprobe repo has added submodules which are not cloned.

The documentation is failing because the RaspberryPi website has changed.

I have attached modified files that appear to correct these issues.

[pico-docs.txt](https://github.com/ndabas/pico-setup-windows/files/10016109/pico-docs.txt)
[pico-setup.txt](https://github.com/ndabas/pico-setup-windows/files/10016110/pico-setup.txt)

I couldn't upload the files with the proper file extensions.

pico-docs.txt should be renamed pico-docs.ps1
pico-setup.txt should be renamed pico-setup.cmd

I was unable to build openocd while building the installer.

ndabas, 2022-11-16:
Thanks for looking into this! Do you think you can create a pull request for these changes, so that you get credited properly in the commit history?

microenh, 2022-11-16:
I’ve never done this before, but I’m happy to try. From what I just read, to create a pull request, I first need to create a branch with the changes. I cloned the repo, created a local branch and committed my two changes to it. When I tried to push the branch back to GitHub, I get a permission error. Mark Sent from Mail for Windows From: Nikhil DabasSent: Wednesday, November 16, 2022 4:09 AMTo: ndabas/pico-setup-windowsCc: Mark Erbaugh; AuthorSubject: Re: [ndabas/pico-setup-windows] Possible Update (Issue #29) Thanks for looking into this! Do you think you can create a pull request for these changes, so that you get credited properly in the commit history?

ndabas, 2022-11-16:
You will need to fork the repo (so you get a copy of the repo in your account), and then edit + push your changes to your fork. Then, you will have the option to create a pull request here using the changes from your fork of the repo. More info: https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request

There's also an 'edit' button on the files which you can see in the GitHub web interface, which will basically fork the repo for you and let you edit the files you want. See https://docs.github.com/en/repositories/working-with-files/managing-files/editing-files#editing-files-in-another-users-repository.

microenh, 2022-11-16:
I have created the pull request.  Thanks for the instructions. I’ve learned something new about git and GitHub. Note: I couldn’t get the build script to build the installer because of issues building openocd, so patched the build script to skip the openocd build, and I copied the openocd files from the current installer in the folders where the openocd build would have put them. Is there an easier way to build the installer building openocd? I didn’t include the updated build script in the pull. Mark Sent from Mail for Windows From: Nikhil DabasSent: Wednesday, November 16, 2022 9:47 AMTo: ndabas/pico-setup-windowsCc: Mark Erbaugh; AuthorSubject: Re: [ndabas/pico-setup-windows] Possible Update (Issue #29) You will need to fork the repo (so you get a copy of the repo in your account), and then edit + push your changes to your fork. Then, you will have the option to create a pull request here using the changes from your fork of the repo. More info: https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-requestThere's also an 'edit' button on the files which you can see in the GitHub web interface, which will basically fork the repo for you and let you edit the files you want. See https://docs.github.com/en/repositories/working-with-files/managing-files/editing-files#editing-files-in-another-users-repository.

ndabas, 2022-11-16:
Thanks! Re: openocd, since picoprobe now supports CMSIS-DAP, we can just use the upstream version instead of the RPi fork. I have scripts which build that correctly, will add those after your patch.

Closing this issue as this is addressed in #30.

closed 2022-11-16 by ndabas