ndabas/pico-setup-windows #31: Can't locate header files

w4mmp, 2023-01-04:
Hello,
I am a new user so I'm sure I'm missing something somewhere but I do have an issue.  
1) Downloaded and installed the latest version of pico-setup-windows. 
2) Via the Pico Project Generator, a new project was created (project name Proficio).
3) Via the Visual Studio Code for Pico, started VS.
4) In VS, selected the new project folder (Proficio).

The Proficio.c that was created by the project generator has the following.  VS reports it can not find the following header files.
#include <stdio.h>
```
#include "pico/stdlib.h"
#include "hardware/spi.h"
#include "hardware/i2c.h"
#include "hardware/dma.h"
#include "hardware/pio.h"
#include "hardware/timer.h"
#include "hardware/clocks.h"

```

Apparently a path is not defined correctly.   What do I need to change so that VS can locate the header files?
Regards,
Ron

w4mmp, 2023-01-04:
Hello,
So, I am attempting to build the basic blink application.
This is what VS reports:
[image]
and

[image]

regards,
Ron

ndabas, 2023-01-05:
Not sure why you concluded that this is a dead project -- it most certainly isn't, with the latest release less than 2 months ago.

If you've moved to an RPi for development -- great. For any future readers who come across this sort of issue and find this thread -- the solution is documented in the [ReadMe.txt](https://github.com/ndabas/pico-setup-windows/tree/master/docs) file included with the installer (and opened by the installer at the end of the installation):

> Visual Studio Code for Pico
Start Visual Studio Code using this link to have the correct environment
variables (like PATH and PICO_SDK_PATH) set for development.

You just need to use the "Visual Studio Code for Pico" shortcut that's copied by the installer into Documents\Pico.

w4mmp, 2023-01-09:
Hello,
Yes, I may have jumped the gun on this,  I apologize.  However your response did not help.  I ran the installer verbatim and did not make any changes.  I let everything default.   And let me say up front I am not experienced with cmake, json file and such.  I simply need to use the IDE to create an application for my needs. 

Regards,
Ron

Then I used the "Pico Project Generator" to produce a simple test project and selected the option to have the generator create the .vs.

Below are screen shots of what is happening.  (Problems are highlighted).

[image]

[image]

[image]

Is it possible that the configuration parameter is case sensitive?
[image]

JDTucker-5696, 2023-01-09:
W4MMP;What solved this for me under Rapian was to open the directory ‘pico-examples’ using VSCode. This seemed to have triggered cmake to configure all the examples with all the appropriate default defines for Intelisense to find the non-system include files.Good luck 🍀 JimT Sent from my iPhone

w4mmp, 2023-01-09:
Hello,
I tried what you suggested on Windows.  No joy.
Regards,
Ron

ndabas, 2023-01-12:
Seems like you need to do two things to make it work:
- For the JSON file error, just change all `\` characters to `\\`. So `"compilerPath": "C:\Program Files (x86)\..."` should become `"compilerPath": "C:\\Program Files (x86)\\..."`
- For the remaining issues, click the `No kit selected` text on the status bar and select the entry with _arm-none-eabi_ if you see it, else choose _Unspecified_. You will see some stuff scroll by in the terminal as CMake configures the project.

Let me know how it goes!

rsemenoff, 2023-01-25:
To be fair, setting up a vscode environment is madness for any platform, and the way it works seems to be a moving target.
One idea is to just to use PIO (which is just basically inline assembly), works in micropython, which because it doesnt
rely on vscode, works seemlessly. Apparently you can write your ISRs in assembly directly within python...

ndabas, 2023-02-23:
Closing this issue as I believe this has now been resolved with the new release from Raspberry Pi.

closed 2023-02-23 by ndabas