ndabas/pico-setup-windows #33: OpenOCD can't find Interface

j-broome, 2023-01-13:
[image]

It's unclear from this repo and the getting started documentation how to properly setup the launch and settings json files. I have PicoProbe running on one Pico with the correct wiring to the Pico I am flashin/debugging. I succeed on a Pi 4, but in Windows 11 I am missing something.

j-broome, 2023-01-13:
The downloads folder shows what following embedded system's (https://learnembeddedsystems.co.uk/133-2) method results in. The "raspberry-pi-swd" is missing from this repo as far as I can tell. Any guidance would be appreciated.

ndabas, 2023-02-23:
Try the new release available from Raspberry Pi: (OpenOCD is also available stand-alone in that release if you do not want the full installation)
https://github.com/raspberrypi/pico-setup-windows/releases/latest

OpenOCD and Picoprobe have been updated in the new release, please take a look at the [tutorial](https://github.com/raspberrypi/pico-setup-windows/blob/master/docs/tutorial.md) and [wiki](https://github.com/raspberrypi/pico-setup-windows/wiki).

Closing this issue for now but feel free to re-open it if there is still an issue.

closed 2023-02-23 by ndabas